ARMmbed/greentea #246: greentea should exit gracefully when key testcase_result['utest_log'] is missing

studavekar, 2017-10-08:
we see this behavior when running tests in parallel ~25 device, This effects when running test in parallel.

root cause :
part of uart logs are missed because of DAPLink Overflow
```
00:06:11.867 [1507470757.66][CONN][RXD] sed, 0 fai<DAPLink:Overflow>
00:06:11.867 [1507470757.66][CONN][INF] found KV pair in stream: {{__testcase_start;DNS literal}}, queued...
00:06:12.034 [1507470757.80][CONN][RXD] case #4: 'DNS preference literal'...
00:06:12.034 [1507470757.80][CONN][RXD] {{__test<DAPLink:Overflow>
00:06:12.034 [1507470757.80][CONN][RXD] reference literal}}
```

here we get parsing error
```
00:06:14.709 mbedgt: checking for GCOV data...
00:06:14.715 mbedgt: mbed-host-test-runner: stopped and returned 'OK'
00:06:14.725 mbedgt: test on hardware with target id: DUMMY
00:06:14.726 mbedgt: test suite 'features-feature_lwip-tests-mbedmicro-net-gethostbyname' ......................... OK in 33.31 sec
00:06:14.727 	test case: 'DNS literal' ..................................................................... ERROR in 0.00 sec
00:06:14.727 	test case: 'DNS preference literal' .......................................................... OK in 0.00 sec
00:06:14.727 	test case: 'DNS preference query' ............................................................ OK in 0.00 sec
00:06:14.728 	test case: 'DNS query' ....................................................................... OK in 0.09 sec
00:06:14.729 mbedgt: test case summary: 4 passes, 0 failures
00:06:14.729 mbedgt: utest test case summary mismatch: utest reported passes and failures miscount!
00:06:14.730 	reported by utest: passes = 4, failures 0)
00:06:14.730 	test case result count: passes = 3, failures 1)
00:06:14.730 mbedgt: checking for 'host_tests' directory above image directory structure
00:06:14.731 	found 'host_tests' directory in: 'TESTS/host_tests'
00:06:14.731 mbedgt: selecting test case observer...
```
Exception trace
```
00:39:48.375 mbedgt: shuffle seed: 0.5876712518
00:39:48.375 mbedgt: exporting to JUNIT file '/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/.axes/target/ARCH_PRO/toolchain/ARM/test_report_ARCH_PRO-ARM.xml'...
00:39:48.854 mbedgt: exporting to HTML file '/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/.axes/target/ARCH_PRO/toolchain/ARM/test_report_ARCH_PRO-ARM.html'...
00:39:48.856 mbedgt: unexpected error:
00:39:48.856 	'utest_log'
00:39:48.857 Traceback (most recent call last):
00:39:48.857   File "/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/.env/bin/mbedgt", line 11, in <module>
00:39:48.857     load_entry_point('mbed-greentea', 'console_scripts', 'mbedgt')()
00:39:48.857   File "/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/greentea/mbed_greentea/mbed_greentea_cli.py", line 412, in main
00:39:48.867     cli_ret = main_cli(opts, args)
00:39:48.867   File "/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/greentea/mbed_greentea/mbed_greentea_cli.py", line 1063, in main_cli
00:39:48.867     html_report = exporter_html(test_report)
00:39:48.867   File "/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/greentea/mbed_greentea/mbed_report_api.py", line 736, in exporter_html
00:39:48.869     test_results)
00:39:48.869   File "/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/greentea/mbed_greentea/mbed_report_api.py", line 624, in get_result_overlay
00:39:48.870     overlay_dropdowns = get_result_overlay_dropdowns(result_div_id, test_results)
00:39:48.870   File "/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/greentea/mbed_greentea/mbed_report_api.py", line 592, in get_result_overlay_dropdowns
00:39:48.870     result_overlay_dropdowns = result_output_dropdown + get_result_overlay_testcases_dropdown_menu(result_div_id, test_results)
00:39:48.870   File "/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/greentea/mbed_greentea/mbed_report_api.py", line 566, in get_result_overlay_testcases_dropdown_menu
00:39:48.870     testcase_results_info += get_result_overlay_testcase_dropdown(result_div_id, index, testcase_result_name, testcase_result)
00:39:48.871   File "/home/arm/jenkins_staging_8_ARCH_PRO/workspace/tm_wrap/61/greentea/mbed_greentea/mbed_report_api.py", line 533, in get_result_overlay_testcase_dropdown
00:39:48.871     "\n".join(testcase_result['utest_log']).rstrip("\n"),
00:39:48.871 KeyError: 'utest_log'
```

Complete logs

[gt_missing_tag_exception.txt](https://github.com/ARMmbed/greentea/files/1366963/gt_missing_tag_exception.txt)

theotherjimmy, 2017-10-08:
Thanks for the logs!

studavekar, 2018-03-05:
closing this issues fix for missing key is been merged.

closed 2018-03-05 by studavekar