ARMmbed/greentea #247: sending more than 45 characters from host test stalls fails

thinkberg, 2017-10-16:
On a NRF52_DK:

I am using a host test as a server side emulation. I need to pass base64 encoded strings back and forth. This works fine for DUT -> HOST, but HOST -> DUT fails if the value is longer than approx. 45 characters. You can use the host test `@event_callback` below to test this. If you remove the last 5-9 characters it works.

```python
    @event_callback("messageLength")
    def __messageLength(self, key, value, timestamp):
        self.send_kv("testLongMessage", "1234567890123456789012345678901234567890123456789")
````

The corresponding Test is:

```C++
void TestLongMessage() {
    char k[48], v[255];
    greentea_send_kv("messageLength", "start");
    greentea_parse_kv(k, v, sizeof(k), sizeof(v));
    greentea_send_kv(k,v);
}
```

Side note: debugging this appeard to read more than 45 characters but at some point only trash was read from the serial, so I cannot say whether it's the python side that might break things.

theotherjimmy, 2017-10-16:
@thinkberg Thanks for the minimized example!

thinkberg, 2017-10-16:
Fail might be the wrong description, the greentea_parse_kv() never returns until the test timeouts.
The same happens if you use a nonsupported character in the value string, like a `|` character.

theotherjimmy, 2017-10-16:
@thinkberg Those cases could be checked on the python side and they could error there.

>The same happens if you use a nonsupported character in the value string

Hmmmm... We should add a check for this, and do something when you try to send "garbage".

thinkberg, 2017-10-16:
What I do now is the following to be able to continue.

host test side:

```python
for pos in xrange(0, len(encodedServerSignedServerMessage), 30):
            self.send_kv("serverSignedServerMessage", encodedServerSignedServerMessage[pos:pos+30])
```

and C++:

```C++
void greentea_parse_kv_slice(char *k, char *v, int keySize, int valueSize, int sliceSize) {
    memset(v, 0, sizeof(v));
    char slice[sliceSize + 1];
    do {
        memset(slice, 0, sizeof(slice));
        greentea_parse_kv(k, slice, keySize, sizeof(slice));
        strcat(v, slice);
    } while (strlen(slice) == sliceSize && strlen(v) < valueSize);
}
```
However, it would be nice if the length issue could either be explicit or fixed.