ARMmbed/greentea #249: Copy method priority incorrect

theotherjimmy, 2017-10-20:
Currently we have priority as follows:
```
internal yotta blob > targets.json > tool default
```
This is a bug.

Instead the priority should be:
```
targets.json /w default > internal yotta blob > tool delaut
```

This implies a few test cases:

In targets.json        | In yotta blob | copy method used | Currently Works
---------------------- | ------------- | ---------------- | ---------------
Yes, with `copy_method`| No            | `targets.json`   | Yes
Yes, without `copy_method`| No            | default   | Yes
Yes, with `copy_method`| Yes           | `targets.json`   | No
Yes, without `copy_method`| Yes           | default   | No
No                     | No           | default       | Yes
No                     | Yes           | yotta blob       | Yes

@bridadan Is this the issue masked by #248?

theotherjimmy, 2017-10-20:
@jeromecoutant @bcostm I prefer to resolve this bug over merging #248

bridadan, 2017-10-20:
@theotherjimmy Perfect 👍

theotherjimmy, 2017-10-20:
@bridadan Does this apply to all target properties? or just copy method?

bridadan, 2017-10-20:
I would prefer it to be for all target properties, we shouldn't be picking and choosing from different sources.

It's currently being used for copy method and reset method

theotherjimmy, 2017-10-20:
I updated the description and title to make it general

PR #250 (2017-10-23): Correct target property priority
And refactor into a bunch more tests.

Fixes #249
Closes #248
closed 2017-10-23 by theotherjimmy