ARMmbed/greentea #251: mbedgt: unexpected error while HTML export

jeromecoutant, 2017-11-03:
My command:
mbedgt -V -v -t NUCLEO_L073RZ-GCC_ARM --report-text=./test_2017_11_03_13_46.txt --report-html=./test_2017_11_03_13_46.html

-> test execution OK
-> text report OK
-> html report crash:

mbedgt: all tests finished!
mbedgt: shuffle seed: 0.0981470632
mbedgt: exporting to TEXT './test_2017_11_03_13_46.txt'...
mbedgt: exporting to HTML file './test_2017_11_03_13_46.html'...
mbedgt: unexpected error:
        'utest_log'
Traceback (most recent call last):
  File "C:\Python27\Scripts\mbedgt-script.py", line 11, in <module>
    load_entry_point('mbed-greentea==1.3.2', 'console_scripts', 'mbedgt')()
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 416, in main
    cli_ret = main_cli(opts, args)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 1067, in main_cli
    html_report = exporter_html(test_report)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_report_api.py", line 755, in exporter_html
    test_results)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_report_api.py", line 644, in get_result_overlay
    overlay_dropdowns = get_result_overlay_dropdowns(result_div_id, test_results)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_report_api.py", line 612, in get_result_overlay_dropdowns
    result_overlay_dropdowns = result_output_dropdown + get_result_overlay_testcases_dropdown_menu(result_div_id, test_results)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_report_api.py", line 584, in get_result_overlay_testcases_dropdown_menu
    testcase_results_info += get_result_overlay_testcase_dropdown(result_div_id, index, testcase_result_name, testcase_result)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_report_api.py", line 551, in get_result_overlay_testcase_dropdown
    "\n".join(testcase_result['utest_log']).rstrip("\n"),
KeyError: 'utest_log'

theotherjimmy, 2017-11-03:
Thanks for the bug report. Is this blocking you?

studavekar, 2018-03-05:
This should be fixed by : https://github.com/ARMmbed/greentea/pull/258

SierpG, 2018-03-12:
I had the same problem with the key time_start. I think in the function get_result_overlay_testcase_dropdown the value for time_start should be somehow initialized. I added the following line  (at line 556 in version 1.3.3) and changed the following line accordingly:

**time_start = testcase_result.get('time_start', testcase_result['time_end'])**
testcase_info = testcase_result_template % (testcase_result['result_text'], testcase_result['duration'],
datetime.datetime.fromtimestamp(**time_star**t).strftime('%d-%m-%Y %H:%M:%S.%f'),
datetime.datetime.fromtimestamp(testcase_result['time_end']).strftime('%d-%m-%Y %H:%M:%S.%f'),
                                                testcase_result['failed'],
                                                testcase_result['passed'],
                                                testcase_utest_log_dropdown)

0xc0170, 2018-06-08:
> This should be fixed by : #258

Anyone to confirm this fixes the bug or not?

jeromecoutant, 2018-06-08:
OK to close this issue

closed 2018-06-08 by jeromecoutant

0xc0170, 2018-08-31:
I believe this should be reopened. Looking at the code, time_start is still not initialized ?

https://github.com/ARMmbed/greentea/blob/bc79ee056a1b41f285201bde35c0a86200a3a535/mbed_greentea/mbed_report_api.py#L728 

Should it be?

theotherjimmy, 2018-08-31:
@0xc0170 That code reference has nothing to do with anything that you said. Perhaps you meant another line?

0xc0170, 2018-09-01:
I remember replying here. the line I highlighted above is the code block that contains the initialization of test results. test results fails with KeyError `time_start` that is accessed later in the chain (I provided a test report in the Jira ticket I opened and this was referenced). This `KeyError` should not happen.