ARMmbed/greentea #256: programming with flash tools that don't enumerate as a MSD

Alex-EEE, 2017-12-08:
Hi @theotherjimmy and @bridadan , thanks again for all the great info yesterday!

It's looking like we can't buy those DIPdaps anymore, so can we revisit the idea of programming an mbed target via a flasher that *does not* enumerate as a mass storage device?  It would be used via the command line.  I know yesterday we were talking about how there would be an issue around target detection, any more thoughts on the direction we should head in trying to work around that?  Happy to back contribute whatever we come up with!

Alex-EEE, 2017-12-08:
Actually, you mentioned this is only an issue in Windows, we could actually run our CI in Windows 10 Ubuntu.  If we did that, what would the trick be to this working?

theotherjimmy, 2017-12-08:
@Alex-EEE The thing we mentioned as a windows only issue was that Mass Storage devices are filtered by vendor. That's only done on windows. You would still need a Mass Storage device on Linux.

bridadan, 2017-12-08:
@Alex-EEE of course! It was great to meet all of you!

I think what we'll need to do is extend the "mock" interface that currently exists: https://github.com/ARMmbed/mbed-ls#mocking-new-or-existing-target-to-custom-platform-name

We will probably need to extend this to include a "persistent" option or something that will make the target always show up as "detected", thus bypassing the current detection functions.

sathysreenath, 2018-02-15:
I was able to to flash the chip using an external debugger (Jlink) and run tests with 'mbed test' with a few changes in both hardware and software as follows:
1) Cut the trace for J12 and J8 jumpers to disconnect flash and serial port. We need to still keep the serial port connected to keep the core code changes to the minimum (this might be an area where we can improve code a bit or use another option if available).
2) Add plugin to the host_test_plugins directory (mbed cli) for the copy (module_copy_pluginname) and reset (module_reset_pluginname) for jlink.
3) Since the trace is cut, we need to explicitly specify the target with --target switch in the mbed command line. Also specify which plugin to use during run option , --copy pluginname  -r pluginname.

Alex-EEE, 2018-12-10:
Here ya go!  I put a modification line in the licenses, let me know if you need anything else for mbed to use it.  Only tested on windows.  Invoke with "--copy jlink  -r jlink"  Credit to @sathysreenath for the code.  Requires JLink commander to be installed on the system.

[module_jlink.zip](https://github.com/ARMmbed/greentea/files/2664048/module_jlink.zip)

bridadan, 2018-12-10:
Oh wow, thanks very much! We don't have an estimate on if/when we'd actually get this into the code base, but I was just curious how you were using the JLink tools for your use case.

Alex-EEE, 2018-12-10:
Cool, well, at least in the meantime it's here for anyone else who might be looking!

bridadan, 2019-01-07:
Related: https://github.com/ARMmbed/mbed-ls/issues/414