ARMmbed/greentea #259: greentea passes the wrong ID when using pyOCD programming on macOS

drewcassidy, 2018-01-22:
when running tests in greentea, mbedhtrun and pyOCD throw errors because it is passing the mbed ID instead of the USB one

```
calling mbedhtrun: mbedhtrun -m OSHCHIP -p /dev/tty.usbmodem14532:9600 -f "BUILD/tests/OSHChip/GCC_ARM/mbed-os/TESTS/mbedmicro-rtos-mbed/heap_and_stack/heap_and_stack.hex" -e "mbed-os/TESTS/host_tests" -d /Volumes/Untitled -C 6 -c pyocd -t 1100021850323120363937383330323133323030ADD5DFD8 -r pyocd --sync 5
```

pyOCD expects the USB ID, which is what it prints with pyocd-gdserver
```
$ pyocd-gdbserver -l                                                  
0 => Arch BLE [nrf51] boardId => 900902211A9C95D4E742FA25
```

theotherjimmy, 2018-01-22:
Yep, that's a bug.

theotherjimmy, 2018-01-22:
I don't currently have time to look at this. Any help would be appreciated.

drewcassidy, 2018-01-22:
Im going to see if I can fix it myself but without other programmers or operating systems to test on I want to make sure I dont break something else in the process. can anyone duplicate the bug on another system?

drewcassidy, 2018-01-26:
It looks like this only happens with older DAPlink firmware versions where the USB and board ID dont match.

bridadan, 2018-01-30:
> It looks like this only happens with older DAPlink firmware versions where the USB and board ID dont match.

Ah yes that makes sense. The good news is recent versions of `mbed-ls` provides both ids in its return data. Sounds like a few modifications to greentea and htrun are necessary to pass this along.

From your first post:
> mbedhtrun throws a bunch of errors about mbed_ls_ext being depricated, so it could also be a dependency issue, though I have the most recent versions of each component installed

That's correct that `mbed_ls_ext` has been deprecated. We just haven't gotten around to updating htrun yet, sorry for the confusion!

The function you'd want to use instead is `list_mbeds`: https://github.com/ARMmbed/mbed-ls/blob/master/mbed_lstools/lstools_base.py#L110

To make sure the correct targetID is found (and therefore the platform name), be sure to call it without changing the default arguments (`list_mbeds()`). This is because `list_mbeds` allows you to speed up the detection process if you aren't as worried about the USB and board ID matching. But in this case, you very much do!

drewcassidy, 2018-01-30:
> That's correct that mbed_ls_ext has been deprecated. We just haven't gotten around to updating htrun yet, sorry for the confusion!

funny you mention that, because I also opened #182 😄

bridadan, 2018-01-30:
And so you have! Nice work! 😄

aashishc1988, 2018-08-14:
@drewcassidy I have started to look at the issue. First, it seems that the only way to reproduce the issue is to downgrade the daplink firmware (havent tried that yet, but I am guessing that should do it). However, after updating daplink, do you still see this issue on any boards/setup?

aashishc1988, 2018-10-02:
@drewcassidy , ping!

drewcassidy, 2018-10-02:
Sorry, I've been away from this project for quite a while working on other things, and I admit I've forgotten a lot about what I was doing...

The problem with updating the daplink is I am using an unusual combination of programmer and target (LPC11U35 and nRF51822), which is not available for the releases for the daplink project. I've been using the firmware images from [this page](https://os.mbed.com/platforms/SWDAP-LPC11U35/) which I have no idea if they are up to date or not. 

is that the cause? is the daplink supposed to be responding with a different value? I was under the impression pyOCD needed the ID of the USB device, which should be independent from the firmware, shouldn't it?

theotherjimmy, 2018-10-02:
@drewcassidy The ID of the USB Device may be controlled by the firmware, as the firmware of the DAPLink is entirely responsible for implementing USB Device (with help from the MCUs PHY).

bridadan, 2018-12-10:
Hey everyone, I'm doing a bit of clean up on issues. From the previous conversation it sounds like the behavior is dependent on DAPLink. Is this an issue you'd still like to work on resolving @drewcassidy? I'd be happy to provide guidance or potentially an initial implementation if you'd be willing to test it!

bridadan, 2019-01-07:
Closing due to inactivity

closed 2019-01-07 by bridadan