ARMmbed/greentea #274: mbedgt 1.4.0 doesn't work on Japanese version of Windows 7

toyowata, 2018-06-07:
### Description

I've got runtime error with mbedgt 1.4.0 on Japanese Windows 7 below:

```
PS C:\Users\toywat01\Documents\GitHub\mbed> mbedgt --version
1.4.0
PS C:\Users\toywat01\Documents\GitHub\mbed> mbed test --run -m rz_a1h -t arm -n tests-mbed_hal-*
mbedgt: greentea test automation tool ver. 1.4.0
mbedgt: test specification file 'C:\Users\toywat01\Documents\GitHub\mbed\BUILD\tests\rz_a1h\arm\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Users\toywat01\Documents\GitHub\mbed\BUILD\tests\rz_a1h\arm\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'RZ_A1H' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: test case filter (specified with -n option)
        tests-mbed_hal-*
        test filtered in 'tests-mbed_hal-rtc_time'
        test filtered in 'tests-mbed_hal-common_tickers'
        test filtered in 'tests-mbed_hal-sleep_manager'
        test filtered in 'tests-mbed_hal-rtc_reset'
        test filtered in 'tests-mbed_hal-critical_section'
        test filtered in 'tests-mbed_hal-ticker'
        test filtered in 'tests-mbed_hal-common_tickers_freq'
        test filtered in 'tests-mbed_hal-rtc_time_conv'
        test filtered in 'tests-mbed_hal-us_ticker'
        test filtered in 'tests-mbed_hal-rtc'
        test filtered in 'tests-mbed_hal-sleep_manager_racecondition'
mbedgt: running 11 tests for platform 'RZ_A1H' and toolchain 'ARM'
mbedgt: mbed-host-test-runner: started
Exception in thread Thread-3:
Traceback (most recent call last):
  File "c:\python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "c:\python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 489, in run_test_thread
    verbose=verbose)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_test_api.py", line 318, in run_host_test
    returncode, htrun_output = run_htrun(cmd, verbose)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_test_api.py", line 134, in run_htrun
    htrun_output += line.decode('utf-8')
  File "c:\python27\lib\encodings\utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x8c in position 10: invalid start byte

mbedgt: could not generate test report
mbedgt: completed in 5.75 sec
mbedgt: exited with code -1000
[mbed] ERROR: "mbedgt" returned error code -1000.
[mbed] ERROR: Command "mbedgt --test-spec C:\Users\toywat01\Documents\GitHub\mbed\BUILD\tests\rz_a1h\arm\test_spec.json -n tests-mbed_hal-*" in "C:\Users\toywat01\Documents\GitHub\mbed"
---
```

### Step to reproduce

* Update latest Mbed greentea
```
$ pip install -U mbed-greentea
```
* Import mbed-os
```
$ mbed import mbed-os
$ cd mbed-os
```
* Run test program
```
$ mbed test -m k64f -t arm
```

### Other information
* There is no error on English version of Windows

cmonr, 2018-06-07:
@toyowata Interesting. The 1.4.0 release was intended to be in lockstep with a new release of htrun in preparation of Python 3 support.

Would you mind running the following commands to check if the issue still persists with the latest changes from htrun?

```
pip uninstall mbed-host-tests
git clone https://github.com/ARMmbed/htrun
pip install -e ./htrun
```

cmonr, 2018-06-08:
@toyowata Interesting. Thanks for the updated log. I'll be aiming to look into it today.

bridadan, 2018-06-08:
@cmonr I have a fix for it up on htrun: https://github.com/ARMmbed/htrun/pull/191

closed 2018-06-08 by theotherjimmy