ARMmbed/greentea #277: Make test suite and test case reports github compatible

jeromecoutant, 2018-07-13:
Test suite and test case reports are formatted like this:

````
+-------------------+---------------+----------------------+--------+--------------------+-------------+
| target            | platform_name | test suite           | result | elapsed_time (sec) | copy_method |
+-------------------+---------------+----------------------+--------+--------------------+-------------+
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_hal-sleep | OK     | 18.3               | default     |
+-------------------+---------------+----------------------+--------+--------------------+-------------+
````

What do you think about replacing "+" by "|" ?
This will ease the copy paste in github comment :-)

````
| target            | platform_name | test suite           | result | elapsed_time (sec) | copy_method |
|-------------------|---------------|----------------------|--------|--------------------|-------------|
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_hal-sleep | OK     | 18.3               | default     |
````

| target            | platform_name | test suite           | result | elapsed_time (sec) | copy_method |
|-------------------|---------------|----------------------|--------|--------------------|-------------|
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_hal-sleep | OK     | 18.3               | default     |

theotherjimmy, 2018-07-13:
I really like this idea.

bridadan, 2018-07-16:
+1 from me!

jeromecoutant, 2018-07-19:
I can push a PR if you indicate to me where this + has to be replaced :-)

theotherjimmy, 2018-07-19:
@jeromecoutant I poked around with doing this in the Mbed OS build tools:
```diff
diff --git a/tools/build_api.py b/tools/build_api.py
index 6f0ef1d49..ca1ee9a58 100644
--- a/tools/build_api.py
+++ b/tools/build_api.py
@@ -1159,7 +1159,7 @@ def mcu_toolchain_matrix(verbose_html=False, platform_filter=None,
     release_version - get the matrix for this major version number
     """
     # Only use it in this function so building works without extra modules
-    from prettytable import PrettyTable
+    from prettytable import PrettyTable, HEADER
     release_version = _lowercase_release_version(release_version)
     version_release_targets = {}
     version_release_target_names = {}
@@ -1181,7 +1181,7 @@ def mcu_toolchain_matrix(verbose_html=False, platform_filter=None,
 
     # All tests status table print
     columns = prepend_columns + unique_supported_toolchains
-    table_printer = PrettyTable(columns)
+    table_printer = PrettyTable(columns, junction_char="|", hrules=HEADER)
     # Align table
     for col in columns:
         table_printer.align[col] = "c"
diff --git a/tools/utils.py b/tools/utils.py
index 640403fe0..22f58a439 100644
--- a/tools/utils.py
+++ b/tools/utils.py
@@ -530,3 +530,7 @@ def integer(maybe_string, base):
         return maybe_string
     else:
         return int(maybe_string, base)
+
+def PrettyTableMarkdown(columns):
+    from prettytable import PrettyTable, HEADER
+    return PrettyTable(columns, junction_char="|", hruls=HEADER)
```

theotherjimmy, 2018-07-19:
Basically, every `PrettyTable` constructor could be replaced with the `PrettyTableMarkdown` that I wrote.

cmonr, 2018-07-23:
👍

jeromecoutant, 2018-08-07:
For information, here is all related PR :

- https://github.com/ARMmbed/mbed-os/pull/7720
- https://github.com/ARMmbed/greentea/pull/279
- https://github.com/ARMmbed/mbed-ls/pull/383
- https://github.com/ARMmbed/htrun/pull/208

closed 2018-09-03 by jeromecoutant