ARMmbed/greentea #286: mbedgt: No space left on device

ReijoSep, 2018-10-31:
Problem: There is not enough disk space for example binary
Environment: Linux Ubuntu 18.04 + NUCLEO-F411RE

Note! Problem is not visible with Virtual box + Ubuntu 18.04 or Windows environment.

It is used an example application from https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/testing/
---------------------
#include "mbed.h"
#include "greentea-client/test_env.h"
#include "unity.h"
#include "utest.h"
#include "rtos.h"
 
using namespace utest::v1;
 
// A test that returns successfully is considered successful
void test_success() {
    TEST_ASSERT(true);
}
 
// Tests that assert are considered failing
void test_failure() {
    TEST_ASSERT(false);
}
 
utest::v1::status_t test_setup(const size_t number_of_cases) {
    // Setup Greentea using a reasonable timeout in seconds
    GREENTEA_SETUP(40, "default_auto");
    return verbose_test_setup_handler(number_of_cases);
}
 
// Test cases
Case cases[] = {
    Case("Testing success test", test_success),
    Case("Testing failure test", test_failure),
};
 
Specification specification(test_setup, cases);
 
// Entry point into the tests
int main() {
    return !Harness::run(specification);
}
-------------------------------------
Steps:
1. blkid:
/dev/sdn: SEC_TYPE="msdos" LABEL="NODE_F411RE" UUID="2702-1974" TYPE="vfat"
2. sudo mount /dev/sdn /media/mbed0
3. mbedls -u
4. sudo mbed test -t GCC_ARM -m RDV_F437VI_GEN1 -DUNIT_TEST=1 -DREPEAT_COUNT=1 -n tests-gen1-example --app-config TESTS/mbed_app_test.json --compile -c -vv

Information:
Package          Version    
---------------- -----------
appdirs          1.4.3      
asn1ate          0.6.0      
asn1crypto       0.24.0     
beautifulsoup4   4.6.3      
certifi          2018.8.24  
cffi             1.11.5     
chardet          3.0.4      
colorama         0.3.9      
cryptography     2.3.1      
ecdsa            0.13       
enum34           1.1.6      
fasteners        0.14.1     
funcsigs         1.0.2      
functools32      3.2.3.post2
future           0.16.0     
futures          3.2.0      
[... 110 lines cut ...]
    result = test_selector.execute()
  File "/usr/local/lib/python2.7/dist-packages/mbed_host_tests/host_tests_runner/host_test_default.py", line 524, in execute
    result = self.mbed.copy_image()
  File "/usr/local/lib/python2.7/dist-packages/mbed_host_tests/host_tests_runner/mbed_base.py", line 187, in copy_image
    initial_remount_count = get_remount_count(disk)
  File "/usr/local/lib/python2.7/dist-packages/mbed_host_tests/host_tests_runner/mbed_base.py", line 100, in get_remount_count
    for line in details_txt.readlines():
IOError: [Errno 5] Input/output error
mbedgt: retry mbedhtrun 1/1
mbedgt: ['mbedhtrun', '-m', u'RDV_F437VI_GEN1', '-p', '/dev/ttyACM0:115200', '-f', u'"BUILD/tests/RDV_F437VI_GEN1/GCC_ARM/TESTS/Gen1/example/example.bin"', '-d', u'/media/mbed0', '-c', 'default', '-t', u'07990221013366613E77F0C5', '-r', 'default', '-C', '4', '--sync', '5', '-P', '60'] failed after 1 count
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test case summary event not found
no test case report present, assuming test suite to be a single test case!
test suite: tests-gen1-example
test case: tests-gen1-example
mbedgt: test on hardware with target id: 07990221013366613E77F0C5
mbedgt: test suite 'tests-gen1-example' .............................................................. TIMEOUT in 5.86 sec
test case: 'tests-gen1-example' .............................................................. ERROR in 5.86 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!

df-command after the problem exists.
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/sdn            2076     2076         0 100% /media/mbed0

theotherjimmy, 2018-10-31:
Hi @ReijoSep. Thanks for the report.

I have edited the comment to put ` ``` ` around the logs, as  using `----------` the line after a block of text will create a title out of the prior line. This was causing your formatting to render incorrectly.

bridadan, 2018-10-31:
Hi @ReijoSep,

The way Mbed platforms work is after a binary has completed flashing, the device will dismount itself, then remount. Windows and Mac take care of this for you automatically, however Linux typically requires you to use an automounter for this.

If the board is never being dismounted, I believe the board's disk usage will eventually fill up, just like you are seeing. Since you included your `mount` command, I'm assuming you're not using an automounter, so that's something you'd need to enable when using greentea.

Let me know if you have further questions!

ReijoSep, 2018-11-06:
Hi @bridadan,

Thanks for Your idea!
Could you tell more about automounter step-by-step, please.
Should I modify the /etc/auto.master file. If YES, how should it be modified?
Should it be modified any other files ( for example /etc/fstab  or any other files)  and how?

My board is NUCLEO-F411RE ST-Link. I would like to see it as like:
/dev/sdc: SEC_TYPE="msdos" LABEL="NODE_F411RE" UUID="2702-1974" TYPE="vfat"

jenkins@nbiot:/etc$ blkid
/dev/sdc: SEC_TYPE="msdos" LABEL="NODE_F411RE" UUID="2702-1974" TYPE="vfat"

jenkins@nbiot:/etc$ mbedls -u
| platform_name | platform_name_unique | mount_point | serial_port  | target_id                | daplink_version |
|---------------|----------------------|-------------|--------------|--------------------------|-----------------|
| unknown       | None[1]              | unknown     | /dev/ttyACM0 | 066EFF494851877267034050 | unknown         |

If I did manual mounting, I will get:

| platform_name  | platform_name_unique | mount_point  | serial_port  | target_id                | daplink_version |
|----------------|----------------------|--------------|--------------|--------------------------|-----------------|
| unknown        | None[0]              | unknown      | /dev/ttyACM1 | 066DFF535353897167061225 | unknown         |
| ST_PLACEHOLDER | ST_PLACEHOLDER[0]    | /media/mbed0 | /dev/ttyACM0 | 07990221013366613E77F0C5 | 0221            |

It will cause fill up the disk, just like you said.

bridadan, 2018-11-06:
Configuring an automounter will be unique to your system. I believe Ubuntu Desktop comes with an automounter, though I haven't used it in quite a while so my knowledge there is pretty stale. I did find this document which may be of use to you: https://help.ubuntu.com/community/AutomaticallyMountPartitions

In the past, I've used a standalone project to handle automounting called [ldm](https://github.com/LemonBoy/ldm). This is a fairly low level tool though, you will need to build it yourself using Make.

ReijoSep, 2018-11-09:
Could it be another problem? It looks that it is not help. After every run : memory is used more and more 
I modified /etc/fstab file. Now it works that if I will disconnect USB cable and reconnect it. Disk mounting works automatically. Have you run Mbed Greentea Test Framework with Ubuntu 18.04 ( without Virtual Machine ) ?

jenkins@nbiot:~/workspace/test_automation/NB-IoT_STM32$ ls -al /dev/disk/by-id/
total 0
drwxr-xr-x 2 root root 240 Nov  9 12:45 .
drwxr-xr-x 7 root root 140 Nov  9 12:52 ..
lrwxrwxrwx 1 root root   9 Nov  9 12:34 ata-hp_HLDS_DVDRW_GUD1N_B7LRVP9534243 -> ../../sr0
lrwxrwxrwx 1 root root   9 Nov  9 12:34 ata-SKhynix_SC300_HFS256G32MND-3210A_ES7BN5180103ABF2W -> ../../sda
lrwxrwxrwx 1 root root  10 Nov  9 12:34 ata-SKhynix_SC300_HFS256G32MND-3210A_ES7BN5180103ABF2W-part1 -> ../../sda1
lrwxrwxrwx 1 root root  10 Nov  9 12:34 ata-SKhynix_SC300_HFS256G32MND-3210A_ES7BN5180103ABF2W-part2 -> ../../sda2
lrwxrwxrwx 1 root root   9 Nov  9 12:34 usb-Generic-_USB3.0_CRW_201408282030-0:0 -> ../../sdb
lrwxrwxrwx 1 root root   9 Nov  9 12:52 usb-MBED_microcontroller_066EFF494851877267034050-0:0 -> ../../sdc
lrwxrwxrwx 1 root root   9 Nov  9 12:34 wwn-0x5001480000000000 -> ../../sr0
lrwxrwxrwx 1 root root   9 Nov  9 12:34 wwn-0x5ace42e020269540 -> ../../sda
lrwxrwxrwx 1 root root  10 Nov  9 12:34 wwn-0x5ace42e020269540-part1 -> ../../sda1
lrwxrwxrwx 1 root root  10 Nov  9 12:34 wwn-0x5ace42e020269540-part2 -> ../../sda2

jenkins@nbiot:~/workspace/test_automation/NB-IoT_STM32$ mbedls -u
WARNING:mbedls.platform_database:Duplicate platform ids found: daplink.0799, ignoring the definitions from /home/jenkins/.local/share/mbms.json
| platform_name   | platform_name_unique | mount_point  | serial_port  | target_id                | daplink_version |
|-----------------|----------------------|--------------|--------------|--------------------------|-----------------|
| RDV_F437VI_GEN1 | RDV_F437VI_GEN1[0]   | /media/mbed0 | /dev/ttyACM0 | 07990221013366613E77F0C5 | 0221            |

jenkins@nbiot:/etc$ cat fstab
UUID=0d38df39-7de5-11e8-bd3c-3c528278fe2a / ext4 defaults 0 0
UUID=24FB-CA77 /boot/efi vfat defaults 0 0
/swap.img       none    swap    sw      0       0
/dev/disk/by-id/usb-MBED_microcontroller_066EFF494851877267034050-0:0 /media/mbed0 vfat auto,rw,user,sync,exec,dev,suid,umask=000 0 0
/dev/disk/by-id/usb-MBED_microcontroller_0675FF333036434B43051516-0:0 /media/mbed1 vfat auto,rw,user,sync,exec,dev,suid,umask=000 0 0

jenkins@nbiot:/etc$ df
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/sdc            2076      356      1720  18% /media/mbed0

bridadan, 2018-11-09:
Hi @ReijoSep, all the behavior you specified are symptoms of the device not dismounting between flashes.

> Have you run Mbed Greentea Test Framework with Ubuntu 18.04 ( without Virtual Machine )

Yup, quite often!

It's possible if you have configuration in your `/etc/fstab` in addition to running an automounter, the automounter may not be able to function properly. I would recommend removing your configuration for the NUCLEO_F411RE from `/etc/fstab`, enable your automounter if not already present, then restart your machine.

Other things to try/questions to ask if you don't have luck above:

1) What automounter are you using? Does it have the correct permissions?
2) Is your NUCLEO board up to date with the latest STLink firmware? The updater is available here: https://www.st.com/en/development-tools/stsw-link007.html

ReijoSep, 2018-11-12:
1. I'm using automount: systemd. Could you tell permissions from your working environment?
Could you compare them for our permission settings, please. Could you send your vfat settings?

https://blog.tomecek.net/post/automount-with-systemd/
https://codingbee.net/tutorials/rhcsa/rhcsa-automounting-using-systemd-and-autofs

jenkins@nbiot:~$ systemctl status media-mbed0.automount
● media-mbed0.automount - mbed0 automounted file system
   Loaded: loaded (/etc/systemd/system/media-mbed0.automount; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-11-12 12:53:55 EET; 21min ago
    Where: /media/mbed0

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
jenkins@nbiot:~$ systemctl status media-mbed0.mount
Warning: The unit file, source configuration file or drop-ins of media-mbed0.mount changed on disk. Run 'systemctl
● media-mbed0.mount - mbed0 disk mounting
   Loaded: loaded (/etc/systemd/system/media-mbed0.mount; disabled; vendor preset: enabled)
   Active: active (mounted) since Mon 2018-11-12 12:53:56 EET; 21min ago
    Where: /media/mbed0
     What: /dev/sdc
  Process: 786 ExecMount=/bin/mount /dev/sdc /media/mbed0 -t vfat -o rw,nosuid,nodev,relatime,uid=1000,gid=1000,dma
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/media-mbed0.mount

Nov 12 12:53:56 nbiot systemd[1]: Mounting mbed0 disk mounting...
Nov 12 12:53:56 nbiot systemd[1]: Mounted mbed0 disk mounting.

jenkins@nbiot:~$ mount | grep sdc
/dev/sdc on /media/mbed0 type vfat (rw,nosuid,nodev,relatime,sync,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
jenkins@nbiot:~$ df
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/sdc            2076        8      2068   1% /media/mbed0

jenkins@nbiot:~$ df
/dev/sdc            2076      124      1952   6% /media/mbed0

jenkins@nbiot:~$ df
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/sdc            2076      240      1836  12% /media/mbed0

jenkins@nbiot:~$ mount | grep sdc
/dev/sdc on /media/mbed0 type vfat (rw,nosuid,nodev,relatime,sync,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)

jenkins@nbiot:~$ cat /etc/systemd/system/media-mbed0.mount
[Unit]
Description=mbed0 disk mounting

[Mount]
What=/dev/sdc
Where=/media/mbed0
Type=vfat
Options=rw,nosuid,nodev,relatime,uid=1000,gid=1000,showexec,utf8,flush,dmask=0000,fmask=0000

[Install]
WantedBy=multi-user.target
jenkins@nbiot:~$ cat /etc/systemd/system/media-mbed0.automount
[Unit]
Description=mbed0 automounted file system

[Automount]
Where=/media/mbed0

[Install]
WantedBy=multi-user.target
jenkins@nbiot:~$

jenkins@nbiot:~$ sudo lsof / | egrep 'PID|\(deleted\)'
COMMAND    PID             USER   FD   TYPE DEVICE  SIZE/OFF     NODE NAME
snapd      908             root    6u   REG    8,2      4096  6946818 /tmp/selftest-squashfs-534518889 (deleted)
jenkins@nbiot:~$

2) Yes, I have the latest firmware of STLink.

bridadan, 2018-11-12:
I don't have a setup ready to go to test this at the moment and I don't think I'll have time to look at it this week, but thanks for all of the steps to reproduce this. Should be invaluable when I do come back to this. I don't really have any other ideas at the moment (short of trying [ldm](https://github.com/LemonBoy/ldm) instead of the systemd automounter you used) unfortunately to unblock you on Linux. If you have access to a Windows machine then that should work. Sorry for the lack-luster answer!

ReijoSep, 2018-11-19:
Please, let me know, if you will try to reproduce my problem or you will need more information about the case.

ReijoSep, 2018-11-29:
Have you any news @bridadan ?

bridadan, 2018-11-29:
Hi @ReijoSep, unfortunately no, quite a bit to do going into the end of the year.

ReijoSep, 2019-01-08:
@bridadan Could you give your answers for my questions ( on 12 Nov 2018 )
I'm using automount: systemd.
1. Could you tell permissions from your working environment?
2. Could you compare them for our permission settings, please.
3. Could you send your vfat settings?
4. Have you tested the latest Mbed Greentea Test Framework with Ubuntu 18.04 ( without Virtual Machine )?

bridadan, 2019-01-09:
Hi @ReijoSep I reproduced this issue. The issue is automount is resetting the permissions of the mount point to root (since systemd runs as root). Greentea then can't do the copying since it doesn't have root permissions. For the purposes of testing, I ran greentea as root (not recommended!), but my device **did fill up like yours is showing**. I tested with a Nucleo F411RE running the very latest stlink firmware (Version V2.J32.M22, which you can download from here: https://www.st.com/en/development-tools/stsw-link007.html).

I don't know enough about systemd and automounting to guide you to solution, sorry about that. I would consult the documentation for those projects.

However, just to be sure, I tested again with using [ldm](https://github.com/LemonBoy/ldm) and it worked great (the disk didn't fill up). So if you want to get started with something that works I'd recommend going that route.

I'm going to close this issue for now since this seems to be an issue with a particular automounter and not greentea.

closed 2019-01-09 by bridadan