ARMmbed/greentea #290: nRF52840_DK platform incorrectly recognized

AGlass0fMilk, 2018-12-07:
I'm trying to run USB device tests as part of my attempt to add USBDevice support to the nRF52840 target on the upcoming USB feature release for Mbed-OS. See issue ARMmbed/mbed-os#7674

The command I am using to run the tests is:
` mbed test -t GCC_ARM -m NRF52840_DK -n mbed-os-tests-usb_device-*`

However, I get the following output (and subsequent error):
```
mbedgt: greentea test automation tool ver. 1.4.0
mbedgt: test specification file './BUILD/tests/NRF52840_DK/GCC_ARM/test_spec.json' (specified with --test-spec option)
mbedgt: using './BUILD/tests/NRF52840_DK/GCC_ARM/test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
[1544150478.92][mbedls.platform_database]Duplicate platform ids found: daplink.0006, ignoring the definitions from /home/gdbeckstein/.local/share/mbedls/platforms.json
mbedgt: detected 1 device
	+---------------+----------------------+--------------+--------------------------+--------------+
	| platform_name | platform_name_unique | serial_port  | mount_point              | target_id    |
	+---------------+----------------------+--------------+--------------------------+--------------+
	| NRF52_DK      | NRF52_DK[0]          | /dev/ttyACM1 | /media/gdbeckstein/JLINK | 000682564699 |
	+---------------+----------------------+--------------+--------------------------+--------------+
mbedgt: processing target 'NRF52840_DK' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
	+---------------+----------------------+-------------+-------------+-----------+
	| platform_name | platform_name_unique | serial_port | mount_point | target_id |
	+---------------+----------------------+-------------+-------------+-----------+
	+---------------+----------------------+-------------+-------------+-----------+
	use 0 instances of execution threads for testing
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.6169567760
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!
mbedgt: completed in 0.01 sec
mbedgt: exited with code -110
[mbed-12271] ERROR: "mbedgt" returned error.
       Code: 146
       Path: "/home/gdbeckstein/Documents/nrf52840-usb-device"
       Command: "mbedgt --test-spec ./BUILD/tests/NRF52840_DK/GCC_ARM/test_spec.json -n mbed-os-tests-usb_device-* -V"
       Tip: You could retry the last command with "-v" flag for verbose output
```
 I have a 3rd-party nRF52840_DK (it's from Rigado, for their BMD340 module based on the nRF52840 since official production DK's were on backorder at the time). The mbedls output is:

```
| platform_name | platform_name_unique | mount_point              | serial_port  | target_id    | daplink_version |
|---------------|----------------------|--------------------------|--------------|--------------|-----------------|
| NRF52_DK      | NRF52_DK[0]          | /media/gdbeckstein/JLINK | /dev/ttyACM1 | 000682564699 | unknown         |
```
So it looks like `mbed test`/`mbedgt` doesn't properly recognize the `NRF52840_DK` target for some reason?

I was able to temporarily fix this by changing the `platform` field in the generated `test_spec.json` file to `NRF52_DK` and removing all write permissions to the file to prevent it from being overwritten. Then the test suite was able to connect and start uploading/running tests as expected.

Am I doing something wrong or is the nRF52840_DK board being incorrectly identified?

j3hill, 2018-12-07:
Please note that the nRF52840_DK works with DAP Link. If you go to the Dap Link firmware page: https://armmbed.github.io/DAPLink/ you can select this platform download the latest firmware file. You can then flash the DAP Link firmware by going into "MAINTENANCE" mode (by holding down the reset button on the nRF52840 board when you plug the USB cable into your development PC), and then dragging the DAP Link firmware file onto the USB "MAINTENANCE" drive. The board will then reboot, and the USB drive will come back up as "DAPLINK."

bridadan, 2018-12-07:
What @j3hill said 😄 That should work. I just checked and the files present on the JLINK drive (what we use to detect which board is which) are the exact same between the NRF52840_DK and the NRF52_DK boards, so `mbed-ls` can't distinguish the two boards. If you use DAPLink, this isn't an issue.

That being said, if you need to use JLink, you should be able to force the tools to pick it up as an NRF52840_DK by using the mocking feature in mbed-ls: https://github.com/ARMmbed/mbed-ls#mocking-renaming-platforms

AGlass0fMilk, 2018-12-07:
@bridadan I would prefer to keep using jlink, as that's what I am familiar with. Perhaps it would be possible to modify the files on the JLINK drive?

I tried to use the `mbedls mock` utility but it didn't work. I'll try it again and let you know what happens.

bridadan, 2018-12-07:
> Perhaps it would be possible to modify the files on the JLINK drive?

I don't believe that will work since the device will override the contents of the file.

Can you post the output of the mock feature? Alternatively I think using "retargeting" should work: https://github.com/ARMmbed/mbed-ls#retargeting-platforms. The features are similar, but retargeting lets you override almost everything. In this case, you'll want to override the `platform_name`.

bridadan, 2019-01-08:
> Oh cool, do you have an ETA for when that changeover is likely to occur?

We're trying to do the move at the end of the week if no one has any objections (on the move PR).

I don't think we can support this out-of-the-box due to the issues stated above (both targets have the same file contents on the mass storage device). For this reason, I'm going to close this. But if you have any further suggestions/questions feel free to reopen it!

closed 2019-01-08 by bridadan