ARMmbed/greentea #294: Bootloader artifacts causes Python 2 crash

screamerbg, 2018-12-18:
This is the traceback I get from the Greentea run:
```
[1545163887.88][CONN][RXD] [BOOT] Application's start address: 0x8010400
[1545163887.88][CONN][RXD] [BOOT] Application's jump address: 0x8011001
[1545163887.88][CONN][RXD] [BOOT] Application's stack address: 0x20020000
[1545163887.88][CONN][RXD] [BOOT] Forwarding to application...
Exception in thread Thread-3:
Traceback (most recent call last):
  File "c:\progra~1\python\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "c:\progra~1\python\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "c:\progra~1\python\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 503, in run_test_thread
    verbose=verbose)
  File "c:\progra~1\python\lib\site-packages\mbed_greentea\mbed_test_api.py", line 326, in run_host_test
    returncode, htrun_output = run_htrun(cmd, verbose)
  File "c:\progra~1\python\lib\site-packages\mbed_greentea\mbed_test_api.py", line 144, in run_htrun
    sys.stdout.write(decoded_line.rstrip() + '\n')
  File "c:\progra~1\python\lib\site-packages\colorama\ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "c:\progra~1\python\lib\site-packages\colorama\ansitowin32.py", line 141, in write
    self.write_and_convert(text)
  File "c:\progra~1\python\lib\site-packages\colorama\ansitowin32.py", line 169, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "c:\progra~1\python\lib\site-packages\colorama\ansitowin32.py", line 174, in write_plain_text
    self.wrapped.write(text[start:end])
  File "c:\progra~1\python\lib\site-packages\colorama\ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "c:\progra~1\python\lib\site-packages\colorama\ansitowin32.py", line 141, in write
    self.write_and_convert(text)
  File "c:\progra~1\python\lib\site-packages\colorama\ansitowin32.py", line 169, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "c:\progra~1\python\lib\site-packages\colorama\ansitowin32.py", line 174, in write_plain_text
    self.wrapped.write(text[start:end])
  File "c:\progra~1\python\lib\encodings\cp866.py", line 12, in encode
    return codecs.charmap_encode(input,errors,encoding_map)
UnicodeEncodeError: 'charmap' codec can't encode character u'\u0600' in position 27: character maps to <undefined>
```

Also here's what the board returns. I assume that the 3 parasitic characters before "mbedmbedmbedmbedmbedmbedmbedmbed" is what causes it.
```
mbed sterm -b 115200
[mbed] Working path "D:\Work\pelion-enablement" (program)
[mbed] Detecting connected targets/boards to your system...
[mbed] Opening serial terminal to "NUCLEO_F207ZG"
--- Terminal on COM29 - 115200,8,N,1 ---
--- [RESET] ---
[BOOT] Mbed Bootloader
[BOOT] ARM: 00000000000000000000
[BOOT] OEM: 00000000000000000000
[BOOT] Layout: 0 80079C8
[BOOT] Active firmware integrity check:
[BOOT] SHA256: 27489FCEE5A3F1A0CBFBC0608C528E8C14E831BC7A54B6571C0AD65D58AC99F8
[BOOT] Version: 1545163709
[BOOT] Slot 0 is empty
[BOOT] Active firmware up-to-date
[BOOT] Application's start address: 0x8010400
[BOOT] Application's jump address: 0x8011001
[BOOT] Application's stack address: 0x20020000
[BOOT] Forwarding to application...
{{__sync;0}}���mbedmbedmbedmbedmbedmbedmbedmbed
{{__sync;0}}
{{__version;1.3.0}}
{{__timeout;600}}
{{__host_test_name;default_auto}}
{{__testcase_count;3}}
>>> Running 3 test cases...
{{__testcase_name;Ethernet network setup}}
{{__testcase_name;Ethernet 1 thread}}
{{__testcase_name;Ethernet 2 threads}}

>>> Running case #1: 'Ethernet network setup'...
{{__testcase_start;Ethernet network setup}}
[NET] IP address is '192.168.137.206'
[NET] MAC address is '00:80:e1:3e:00:2d'
{{__testcase_finish;Ethernet network setup;1;0}}
>>> 'Ethernet network setup': 1 passed, 0 failed

>>> Running case #2: 'Ethernet 1 thread'...
{{__testcase_start;Ethernet 1 thread}}
[NET-1] Registered socket callback function
[NET-1] Request header (66): GET /mbed-test-files/alice.txt  HTTP/1.1
Host: armmbed.github.io

[NET-1] Non-blocking socket mode set
[NET-1] Found body index: 622
[NET-1] Received bytes: 398, (0.24%, 9.41KB/s, ETA: 00:00:17)
[NET-1] Received bytes: 722, (0.43%, 15.06KB/s, ETA: 00:00:10)
[NET-1] Received bytes: 1518, (0.91%, 28.20KB/s, ETA: 00:00:05)
[NET-1] Received bytes: 2542, (1.52%, 16.11KB/s, ETA: 00:00:10)
[NET-1] Received bytes: 3566, (2.13%, 21.79KB/s, ETA: 00:00:07)
[NET-1] Received bytes: 3662, (2.19%, 21.61KB/s, ETA: 00:00:07)
[NET-1] Received bytes: 4686, (2.80%, 17.70KB/s, ETA: 00:00:08)
[NET-1] Received bytes: 5122, (3.06%, 18.92KB/s, ETA: 00:00:08)
```

My setup:
```
$ mbed --version
1.8.3

$ mbedgt --version
1.5.0

$ mbedhtrun --version
1.4.4

```

bridadan, 2018-12-18:
Huh, how very strange! We decode the line right before the traceback you posted: https://github.com/ARMmbed/greentea/blob/master/mbed_greentea/mbed_test_api.py#L134

I'll be honest, I'm grasping at straws here. However the deep colorama traceback makes me curious. Could you try the following?

1. What version of colorama do you have installed:
```
$ pip show colorama
Name: colorama
Version: 0.3.9
Summary: Cross-platform colored terminal text.
Home-page: https://github.com/tartley/colorama
Author: Arnon Yaari
Author-email: tartley@tartley.com
License: BSD
Location: c:\python27\lib\site-packages
Requires:
Required-by: pytest, pyocd, pylint, manifest-tool
```

2. Try updating it: `pip install -U colorama`
3. Try running it again

screamerbg, 2018-12-18:
@bridadan Thanks for the fast response.

This is the colorama version.

```
$ pip show colorama

Name: colorama
Version: 0.3.9
Summary: Cross-platform colored terminal text.
Home-page: https://github.com/tartley/colorama
Author: Arnon Yaari
Author-email: tartley@tartley.com
License: BSD
Location: c:\progra~1\python\lib\site-packages
Requires:
Required-by: pytest, pyocd, pylint, mbed-greentea, manifest-tool
```

0xc0170, 2018-12-19:
Note, I remember seeing something similar in the CI logs. If the serial communication is not working correctly, some characters are incorrectly received as show above, this leads to tracebacks. 
If I come back to this issue again in CI tests, I'll share it, do not have at the moment particular job.

screamerbg, 2018-12-23:
Bump @bridadan this is blocking normal testing in our test system. Could you please take a look at it? Note that it's not an htrun issue as removing the `-v` flag doesn't cause greentea to crash. Unfortunately, we need `-v` so we're able to report detailed issues to partners.

bridadan, 2018-12-23:
Hey there, sorry I'm out for the holidays until the new year (not near any hardware). However I had a thought regarding your issue: is your terminal set to utf-8? On Windows it looks like you can run a command to change it: https://stackoverflow.com/questions/388490/how-to-use-unicode-characters-in-windows-command-line/388500#388500

I realize this may not be the solution (though if it works, it could be used as a workaround), I'm more trying to figure out what's causing the issue.

screamerbg, 2019-01-04:
@bridadan Happy New Year.

As a follow up, perhaps one of the easiest ways to reproduce is to create a simple test application which prints unicode characters.

```
[1546616905.27][CONN][RXD] [BOOT] Forwarding to application...
Exception in thread Thread-41:
Traceback (most recent call last):
  File "d:\python\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "d:\python\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "d:\python\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 489, in run_test_thread
    verbose=verbose)
  File "d:\python\lib\site-packages\mbed_greentea\mbed_test_api.py", line 318, in run_host_test
    returncode, htrun_output = run_htrun(cmd, verbose)
  File "d:\python\lib\site-packages\mbed_greentea\mbed_test_api.py", line 143, in run_htrun
    sys.stdout.write(line.decode('utf-8').rstrip() + '\n')
  File "d:\python\lib\site-packages\colorama\ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "d:\python\lib\site-packages\colorama\ansitowin32.py", line 141, in write
    self.write_and_convert(text)
  File "d:\python\lib\site-packages\colorama\ansitowin32.py", line 169, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "d:\python\lib\site-packages\colorama\ansitowin32.py", line 174, in write_plain_text
    self.wrapped.write(text[start:end])
  File "d:\python\lib\site-packages\colorama\ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "d:\python\lib\site-packages\colorama\ansitowin32.py", line 141, in write
    self.write_and_convert(text)
  File "d:\python\lib\site-packages\colorama\ansitowin32.py", line 169, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "d:\python\lib\site-packages\colorama\ansitowin32.py", line 174, in write_plain_text
    self.wrapped.write(text[start:end])
  File "d:\python\lib\encodings\cp850.py", line 12, in encode
    return codecs.charmap_encode(input,errors,encoding_map)
UnicodeEncodeError: 'charmap' codec can't encode character u'\u0600' in position 27: character maps to <undefined>
```

`\u0600` seems like a good "candidate" for unicode crash trigger :)

bridadan, 2019-01-07:
Hi @screamerbg, thanks for providing more info. I'm having a heck of a time reproducing this issue. I've tried to write a unit test to reproduce this, but I haven't had any luck creating the problem in either Python 2 or Python 3. 

Here's the branch to my unit test so you can try it on your machine (branch name is "encode_failure"): https://github.com/bridadan/greentea/tree/encode_failure

Looks like you're using this respository: https://github.com/ARMmbed/pelion-enablement. Is there a specific commit I should use to try and reproduce the failure?

bridadan, 2019-01-08:
I _believe_ I've found the issue. Can you try this branch and see if it works? https://github.com/bridadan/greentea/tree/encode_failure

bridadan, 2019-01-09:
@jeromecoutant reported a similar issue and has reported that the fix works for him.

PR #297 (2019-01-11): Fix encode failure
This should fix #294. Certain characters (typically triggered by corrupted data coming over a device's serial line) can cause greentea to crash when printing them to the console. This seems to only be an issue with Python 2.7. Handling this case universally between Python 2.7 and 3.x turns out to be quite challenging and I didn't really find any solutions I was completely happy with. This was the best compromise I could find.

Python 2.7 is perfectly happy to accept "byte-like" strings when writing to stdout. This also fixes the traceback described in #294. Python 3.x howeve [... cut]
closed 2019-01-11 by theotherjimmy