ARMmbed/greentea #296: Unable execute any test case using greentea command

UrmilaKL, 2019-01-09:
If i try to execute any testcase using below command it is failing with below error, I have  created dummy main.cpp in TESTS\Group1\TestCase1\ and tried to execute it
C:\mbedOSExample\mbed-os-example-blinky>mbedgt -n tests* -t GCC_ARM -m DISCO_L475VG_IOT01A
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: using multiple test specifications from current directory!
        using 'BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
[1547031484.20][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
mbedgt: detected 1 device
mbedgt: shuffle seed: 0.1498640650
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!
mbedgt: completed in 0.96 sec
mbedgt: exited with code -110

C:\mbedOSExample\mbed-os-example-blinky>mbedhtrun --version
1.4.4

C:\mbedOSExample\mbed-os-example-blinky>mbedgt -l
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: using multiple test specifications from current directory!
        using 'BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json'
mbedgt: available tests for built 'DISCO_L475VG_IOT01A-GCC_ARM', location 'BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM'

bridadan, 2019-01-09:
Hi @UrmilaKL,

It looks like you may have your commands confused, you're using `mbedgt` instead of `mbed`. Try running this command instead:

```
mbed -n tests* -t GCC_ARM -m DISCO_L475VG_IOT01A
```

You're also running a pretty old version of greentea so you may want to update that.

UrmilaKL, 2019-01-09:
Hi @bridadan,

After upgrading greenTea command to 1.5.0 i tried executing the below command 

mbed test -n tests* -t GCC_ARM -m DISCO_L475VG_IOT01A

I got below error, Kindly let me know if anything i missed to add

mbedgt: test specification file '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'DISCO_L475VG_IOT01A' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: test case filter (specified with -n option)
        tests*
mbedgt: invalid test case names (specified with '-n' option)
mbedgt: test name 'tests*' not found in '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' (specified with --test-spec option)
        note: test case names are case sensitive
        note: see list of available test cases below
mbedgt: available tests for build 'DISCO_L475VG_IOT01A-GCC_ARM', location 'BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM'
mbedgt: running 0 tests for platform 'DISCO_L475VG_IOT01A' and toolchain 'GCC_ARM'
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.3913133991
mbedgt: no platform/target matching tests were found!
mbedgt: completed in 5.00 sec
mbedgt: exited with code -10
[mbed] ERROR: "mbedgt" returned error.
       Code: -10
       Path: "C:\mbedOSExample\mbed-os-example-blinky"
       Command: "mbedgt --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json -n tests*"
       Tip: You could retry the last command with "-v" flag for verbose output

bridadan, 2019-01-09:
@UrmilaKL if you could wrap your logs and commands in code blocks, that would help quite a bit :) I've added them to your above post.

Looking at this section of the log:

```
mbedgt: test case filter (specified with -n option)
        tests*
mbedgt: invalid test case names (specified with '-n' option)
mbedgt: test name 'tests*' not found in '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' (specified with --test-spec option)
        note: test case names are case sensitive
        note: see list of available test cases below
mbedgt: available tests for build 'DISCO_L475VG_IOT01A-GCC_ARM', location 'BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM'
```

It looks like you don't have any test built? Do you have the full logs from your above command (including the build step)?

Having the output of this would also be helpful:
```
mbed test -n tests* -t GCC_ARM -m DISCO_L475VG_IOT01A --run-list
```

UrmilaKL, 2019-01-10:
Hi Brindadan,
Please find the output of the below command,
C:\mbedOSExample\mbed-os-example-blinky>mbed test -n tests* -t GCC_ARM -m DISCO_L475VG_IOT01A --run-list
[mbed] Working path "C:\mbedOSExample\mbed-os-example-blinky" (program)
mbedgt: greentea test automation tool ver. 1.5.0
mbedgt: test specification file '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' from current directory!
mbedgt: available tests for build 'DISCO_L475VG_IOT01A-GCC_ARM', location 'BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM'

UrmilaKL, 2019-01-10:
@ciarmcom ,

I am unable to access this JIRA, May be i don't have access to this link

My test cases are getting failed because of sync_failed issue, Please find my test case below

C:\mbedOSExample\mbed-os-example-blinky>mbedgt --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json -n tests-Group1-TestCase3
mbedgt: greentea test automation tool ver. 1.5.0
mbedgt: test specification file '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'DISCO_L475VG_IOT01A' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: test case filter (specified with -n option)
        tests-Group1-TestCase3
        test filtered in 'tests-group1-testcase3'
mbedgt: running 1 test for platform 'DISCO_L475VG_IOT01A' and toolchain 'GCC_ARM'
mbedgt: mbed-host-test-runner: started
mbedgt: retry mbedhtrun 1/1
mbedgt: ['mbedhtrun', '-m', u'DISCO_L475VG_IOT01A', '-p', u'COM6:9600', '-f', u'"BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/TESTS/Group1/TestCase3/TestCase3.bin"', '-d', 'E:', '-c', 'default', '-t', u'076402211830616C2774F735', '-r', 'default', '-C', '4', '--sync', '5', '-P', '60'] failed after 1 count
mbedgt: checking for GCOV data...
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-group1-testcase3
        test case: tests-group1-testcase3
mbedgt: test on hardware with target id: 076402211830616C2774F735
mbedgt: test suite 'tests-group1-testcase3' .......................................................... SYNC_FAILED in 38.25 sec
        test case: 'tests-group1-testcase3' .......................................................... ERROR in 38.25 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.8400194244
mbedgt: test suite report:
| target                      | platform_name       | test suite             | result      | elapsed_time (sec) | copy_method |
|-----------------------------|---------------------|------------------------|-------------|--------------------|-------------|
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | tests-group1-testcase3 | SYNC_FAILED | 38.25              | default     |
mbedgt: test suite results: 1 SYNC_FAILED
mbedgt: test case report:
| target                      | platform_name       | test suite             | test case              | passed | failed | result | elapsed_time (sec) |
|-----------------------------|---------------------|------------------------|------------------------|--------|--------|--------|--------------------|
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | tests-group1-testcase3 | tests-group1-testcase3 | 0      | 1      | ERROR  | 38.25              |
mbedgt: test case results: 1 ERROR
mbedgt: completed in 43.82 sec
mbedgt: exited with code 1

C:\mbedOSExample\mbed-os-example-blinky>mbedgt --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json -n tests-Group1-TestCase3
mbedgt: greentea test automation tool ver. 1.5.0
mbedgt: test specification file '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'DISCO_L475VG_IOT01A' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: test case filter (specified with -n option)
        tests-Group1-TestCase3
        test filtered in 'tests-group1-testcase3'
mbedgt: running 1 test for platform 'DISCO_L475VG_IOT01A' and toolchain 'GCC_ARM'
mbedgt: mbed-host-test-runner: started
mbedgt: retry mbedhtrun 1/1
mbedgt: ['mbedhtrun', '-m', u'DISCO_L475VG_IOT01A', '-p', u'COM6:9600', '-f', u'"BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/TESTS/Group1/TestCase3/TestCase3.bin"', '-d', 'E:', '-c', 'default', '-t', u'076402211830616C2774F735', '-r', 'default', '-C', '4', '--sync', '5', '-P', '60'] failed after 1 count
mbedgt: checking for GCOV data...
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-group1-testcase3
        test case: tests-group1-testcase3
mbedgt: test on hardware with target id: 076402211830616C2774F735
mbedgt: test suite 'tests-group1-testcase3' .......................................................... SYNC_FAILED in 38.14 sec
        test case: 'tests-group1-testcase3' .......................................................... ERROR in 38.14 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.8694350299
mbedgt: test suite report:
| target                      | platform_name       | test suite             | result      | elapsed_time (sec) | copy_method |
|-----------------------------|---------------------|------------------------|-------------|--------------------|-------------|
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | tests-group1-testcase3 | SYNC_FAILED | 38.14              | default     |
mbedgt: test suite results: 1 SYNC_FAILED
mbedgt: test case report:
| target                      | platform_name       | test suite             | test case              | passed | failed | result | elapsed_time (sec) |
|-----------------------------|---------------------|------------------------|------------------------|--------|--------|--------|--------------------|
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | tests-group1-testcase3 | tests-group1-testcase3 | 0      | 1      | ERROR  | 38.14              |
mbedgt: test case results: 1 ERROR
mbedgt: completed in 44.08 sec
mbedgt: exited with code 1

bridadan, 2019-01-10:
Hi @UrmilaKL, you can ignore the ciarmcom account, that's just a bot account we use to track issues within Arm.

Glad to see you have tests being flashed to the device now! To see why the sync is failing, I'll need the verbose log:

```
mbedgt --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json -n tests-Group1-TestCase3 -V
```

Notice I've added the `-V` option to your command, could you please post the full log of that command?

bridadan, 2019-01-10:
Also, I found this note on the board's mbed page: https://os.mbed.com/platforms/ST-Discovery-L475E-IOT01A/#known-limitations

If you have revision C of the board, you may not be able to use it with greentea (since greentea resets the board progma

UrmilaKL, 2019-01-11:
Hi @bridadan ,

Please find the below logs with versbose option,

C:\mbedOSExample\mbed-os-example-blinky>mbedgt --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json -n tests-Group1-TestCase1 -V
mbedgt: greentea test automation tool ver. 1.5.0
mbedgt: test specification file '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using '.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        | platform_name       | platform_name_unique   | serial_port | mount_point | target_id                |
        |---------------------|------------------------|-------------|-------------|--------------------------|
        | DISCO_L475VG_IOT01A | DISCO_L475VG_IOT01A[0] | COM6        | E:          | 076402211830616C2774F735 |
mbedgt: processing target 'DISCO_L475VG_IOT01A' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        | platform_name       | platform_name_unique   | serial_port | mount_point | target_id                |
        |---------------------|------------------------|-------------|-------------|--------------------------|
        | DISCO_L475VG_IOT01A | DISCO_L475VG_IOT01A[0] | COM6:9600   | E:          | 076402211830616C2774F735 |
mbedgt: test case filter (specified with -n option)
        tests-Group1-TestCase1
        test filtered in 'tests-group1-testcase1'
mbedgt: running 1 test for platform 'DISCO_L475VG_IOT01A' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found: two directory levels above image path checked
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m DISCO_L475VG_IOT01A -p COM6:9600 -f "BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/TESTS/Group1/TestCase1/TestCase1.bin" -d E: -c default -t 076402211830616C2774F735 -r default -C 4 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1547178952.23][HTST][INF] host test executor ver. 1.4.4
[1547178952.23][HTST][INF] copy image onto target...
[1547178952.23][COPY][INF] Waiting up to 60 sec for '076402211830616C2774F735' mount point (current is 'E:')...
        1 file(s) copied.
[1547178962.09][HTST][INF] starting host test process...
[1547178965.19][CONN][INF] starting connection process...
[1547178965.19][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1547178965.19][CONN][INF] initializing serial port listener...
[1547178965.19][PLGN][INF] Waiting up to 60 sec for '076402211830616C2774F735' serial port (current is 'COM6')...
[1547178965.21][HTST][INF] setting timeout to: 60 sec
[1547178965.26][SERI][INF] serial(port=COM6, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1547178965.26][SERI][INF] reset device using 'default' plugin...
[1547178965.52][SERI][INF] waiting 1.00 sec after reset
[1547178966.52][SERI][INF] wait for it...
[1547178966.55][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1547178966.55][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1547178966.55][CONN][INF] sending preamble '3643d17c-2e69-4625-96f5-5c40477336da'
[1547178966.60][SERI][TXD] {{__sync;3643d17c-2e69-4625-96f5-5c40477336da}}
[1547178966.62][CONN][RXD] =============================== SYSTEM INFO  ================================
[1547178966.62][CONN][RXD] Mbed OS Version: 51101
[1547178966.62][CONN][RXD] CPU ID: 0x410fc241
[1547178966.62][CONN][RXD] Compiler ID: 2
[1547178966.62][CONN][RXD] Compiler Version: 60300
[1547178966.62][CONN][RXD] ================= CPU STATS =================
[1547178966.62][CONN][RXD] Idle: 4% Usage: 96%
[1547178966.62][CONN][RXD] ================ HEAP STATS =================
[1547178966.62][CONN][RXD] Current heap: 1236
[1547178966.63][CONN][RXD] Max heap size: 1236
[1547178966.63][CONN][RXD] ================ THREAD STATS ===============
[1547178966.63][CONN][RXD] ID: 0x20001c3c
[1547178966.63][CONN][RXD] Name: main
[1547178966.63][CONN][RXD] State: 2
[1547178966.63][CONN][RXD] Priority: 24
[1547178966.63][CONN][RXD] Stack Size: 4096
[1547178966.63][CONN][RXD] Stack Space: 3064
[1547178966.63][CONN][RXD] ID: 0x20001494
[1547178966.63][CONN][RXD] Name: rtx_idle
[1547178966.63][CONN][RXD] State: 1
[1547178966.63][CONN][RXD] Priority: 1
[1547178966.64][CONN][RXD] Stack Size: 768
[1547178966.64][CONN][RXD] Stack Space: 256
[1547178966.65][CONN][RXD] ID: 0x20001450
[1547178966.66][CONN][RXD] Name: rtx_timer
[... 153 lines cut ...]
mbedgt: mbed-host-test-runner: stopped and returned 'SYNC_FAILED'
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-group1-testcase1
        test case: tests-group1-testcase1
mbedgt: test on hardware with target id: 076402211830616C2774F735
mbedgt: test suite 'tests-group1-testcase1' .......................................................... SYNC_FAILED in 43.37 sec
        test case: 'tests-group1-testcase1' .......................................................... ERROR in 43.37 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.6498896417
mbedgt: test suite report:
| target                      | platform_name       | test suite             | result      | elapsed_time (sec) | copy_method |
[... cut ...]
| target                      | platform_name       | test suite             | test case              | passed | failed | result | elapsed_time (sec) |
|-----------------------------|---------------------|------------------------|------------------------|--------|--------|--------|--------------------|
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | tests-group1-testcase1 | tests-group1-testcase1 | 0      | 1      | ERROR  | 43.37              |
mbedgt: test case results: 1 ERROR
mbedgt: completed in 49.32 sec
mbedgt: exited with code 1

C:\mbedOSExample\mbed-os-example-blinky>

UrmilaKL, 2019-01-11:
@bridadan ,

Thanks for the information,
I am using Rev B board for my programming

bridadan, 2019-01-11:
Oh I'm sorry, I didn't mean to close this issue!

Ok so the output from that log tells me that the program present on the board is not a valid test case. In fact, it looks like its the blinky example found here: https://github.com/ARMmbed/mbed-os-example-blinky.

Is the device being programmed correctly? Greentea uses the USB mass storage device of the STLink to program the board and then the target is reset by sending a serial break over the USB serial port. Does this work for this revision of the board?

bridadan, 2019-01-11:
Hi @UrmilaKL, just a heads up I'm moving this issue to the https://github.com/armmbed/mbed-os-tools repository with the rest of the greentea project today. We can continue the conversation over there (GitHub should handle all of the link redirects).