ARMmbed/greentea #298: Issue for the colorama's version

heejungPark, 2019-01-10:
We are trying to test pelion-enablement.
But the command, _mbed test -t ARM -m SDT64B -n simple-mbed-cloud-client-tests-* --run -v_, is typed, the error is occurred like below.
Our colorama's version is 0.4.1. So we think it caused this error.
Is the only solution for this error lower colorama's version than 0.4.0??

```
C:\mbed\Workspace\Test\pelion-enablement>mbed test -t ARM -m SDT64B -n simple-mbed-cloud-client-tests-* --run -v
[mbed] Working path "C:\mbed\Workspace\Test\pelion-enablement" (program)
[mbed] Exec "mbedgt --test-spec .\BUILD\tests\SDT64B\ARM\test_spec.json -n simple-mbed-cloud-client-tests-* -V" in "C:\mbed\Workspace\Test\pelion-enablement"
Traceback (most recent call last):
  File "C:\mbed\Tools\mbed-CLI\Python27\Scripts\mbedgt-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 3088, in <module>
    @_call_aside
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 3072, in _call_aside
    f(*args, **kwargs)
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 3101, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 576, in _build_master
    return cls._build_from_requirements(__requires__)
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 589, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 778, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'colorama<0.4,>=0.3' distribution was not found and is required by mbed-greentea
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "C:\mbed\Workspace\Test\pelion-enablement"
       Command: "mbedgt --test-spec .\BUILD\tests\SDT64B\ARM\test_spec.json -n simple-mbed-cloud-client-tests-* -V"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

bridadan, 2019-01-10:
Hi @heejungPark,

This was fixed in version 1.5.0 of greentea, please upgrade with the following command: `pip install -U mbed-greentea`.

bridadan, 2019-01-11:
Hi @heejungPark, the mismatch in the versions seems to be caused by older version of the `requests` library. Can you run `pip install -U requests` and try again?

bridadan, 2019-01-11:
Hi @heejungPark , just a heads up I'm moving this issue to the https://github.com/armmbed/mbed-os-tools repository with the rest of the greentea project today. We can continue the conversation over there (GitHub should handle all of the link redirects).