ARMmbed/greentea #299: TypeError: cannot use a string pattern on a bytes-like object

jupe, 2019-01-10:
Greentea crashes in macOS with python 3.6. See logs:

```
$ mbedgt
mbedgt: greentea test automation tool ver. 1.4.0
mbedgt: using multiple test specifications from current directory!
	using 'BUILD/tests/K64F/GCC_ARM/test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'K64F' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: running 92 tests for platform 'K64F' and toolchain 'GCC_ARM'
mbedgt: mbed-host-test-runner: started
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mbed_greentea/mbed_greentea_cli.py", line 489, in run_test_thread
    verbose=verbose)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mbed_greentea/mbed_test_api.py", line 318, in run_host_test
    returncode, htrun_output = run_htrun(cmd, verbose)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mbed_greentea/mbed_test_api.py", line 138, in run_htrun
    test_error = htrun_failure_line.search(line)
TypeError: cannot use a string pattern on a bytes-like object

mbedgt: could not generate test report
mbedgt: completed in 3.24 sec
mbedgt: exited with code -1000
```

bridadan, 2019-01-10:
Hi @jupe, looks like the version you're using is v1.4.0. From your log:

```
mbedgt: greentea test automation tool ver. 1.4.0
```

Can you upgrade to the latest version and give it a try? `pip install -U mbed-greentea`

jupe, 2019-01-10:
yep have to try with python 3 and mbedgt@1.5.0 . Do we have CI tests with this combination?

bridadan, 2019-01-10:
We don't have coverage of this part of the code base quite yet, we're working to add it in https://github.com/armmbed/mbed-os-tools.

bridadan, 2019-01-10:
@jupe I'm pretty confident this isn't an issue in greentea as of version 1.5.0 (patched here: https://github.com/ARMmbed/greentea/pull/292)

bridadan, 2019-01-10:
@jupe I'll go ahead and close this, can please reopen this if you run into the issue with 1.5.0?

closed 2019-01-10 by bridadan