ARMmbed/greentea #303: Make greentea standalone

rwalton-arm, 2021-05-25:
We'd like to make the greentea tools standalone again, rather than part of the mbed-os-tools mono repo. 

- [ ] Separate Greentea host tools (python) into their own repo 
- [ ] Bring mbed_host_tests (htrun) into standlone greentea python library
- [ ] Document how to implement your own greentea host tests
- [ ] Set up Greentea host tools PR testing (GitHub Actions)
- [ ] Create a standalone mbed-host-tests package for mbed-specific host tests (or part of greentea package)
- [ ] Refactor Mbed host-tests to use standalone greentea python lib
- [ ] Set up Greentea host tools continuous deployment pypi releases (GitHub Actions)
- [ ] Document how to generate test results and interpret results

LDong-Arm, 2021-06-08:
Explanation for scripts in [host_tests](https://github.com/ARMmbed/greentea/tree/65c97b6c8392a0072dcd5f7914d798d7357ee7f2/src/htrun/host_tests):
Some of the scripts are not directly referenced by the code base (e.g. `default_auto.py`), but actually greentea-client can instruct htrun to run the script with specified name:
```c
GREENTEA_SETUP(20, "default_auto");
```
The `host_tests` from htrun itself is in the "search path", but users can also add additional directories using the `-e/--enum-host-tests` option. Also, mbed-os [bundles its own host tests](https://github.com/ARMmbed/mbed-os/tree/mbed-os-6.11.0/tools/host_tests).

rajkan01, 2021-06-14:
Verify fastmodel support works in standalone greentea htrun: 

`mbed-fastmodel-agent` is an interface python module to ARM fast model and this fastmodel agent is invoked by `htrun` to run host tests with fast models. Unfortunately `mbed-fastmodel-agent` is using pyCADI (based Python 2.7) which is part of the ARM fast model and it got deprecated in the latest version and replaced by Irish CADI which leads to significant changes in the `mbed-fastmodel-agent` to work so I could not able to completely verify the flow of fast model inside htrun and only checked ran the following command to make sure it invokes the fast model agent APIs

Patater, 2021-09-30:
All tasks closed. Mission complete.

closed 2021-09-30 by Patater