ARMmbed/greentea #317: Use mbed-tools detect --json for device detection, to avoid needing to depend on any Python library for device detection

rwalton-arm, 2021-07-09:


rajkan01, 2021-07-19:
As we decided to use htrun with CTest, this issue proposed "mbed-tools detect --json" command-line argument is no longer required, so closing this issue

rwalton-arm, 2021-07-19:
This should still be relevant. `htrun`, CTest and CMake don't provide a way to do device detection, so we need to pass in mount point, serial port, etc... on the CMake command line. `mbedtools detect --json` would be one way of detecting the device. I don't understand the edited description to be honest, this would be _irrelevant_ if we _did_ decide to use the mbedgt command line (as that gave us a way to detect devices) but we aren't using it, so we probably need another way to detect the device.

rajkan01, 2021-07-19:
This issue needs further clarification as CMake and CTest don't provide support for device detection.

rajkan01, 2021-07-21:
As with the new approach greentea test htrun command run by CTest instead of mbedgt so we no longer required this proposed command-line argument and closing this issue

closed 2021-07-21 by rajkan01